lewisMW/agriculture-SoC #10: Timer IP Specifications

lewisMW, 2025-01-15:
- [ ] Create specs and do some research for the timer IP. Fill in scaffold here: https://unisydneyedu-my.sharepoint.com/:w:/g/personal/lwat0045_uni_sydney_edu_au/EZJQlrRwf0pDgi94zrjHnGgBlbid-Y7o1PqV0c9euZl2bg?e=o8822p (ChatGPT generated, feel free to delete stuff).
- [ ] Implement OR find IP that fulfill the spec (should be many, maybe ARM, or ChatGPT, etc).
- [ ] Create simple unit test (ChatGPT is easy!)

pgaft, 2025-01-28:
Found the PL031 RTC which is a suitable IP for this use case. Since we are currently unable to access AAA IP, a dummy RTC has been implemented on the adc-apb-wrapper branch.

lewisMW, 2025-02-02:
@pgaft I've just checked and we should have access to https://developer.arm.com/downloads/view/PL031?sortBy=availableBy&revision=r1p3-00rel0-2 now.

closed 2025-04-04 by lewisMW