lewisMW/agriculture-SoC #11: Analog-Sensor Requirements

lewisMW, 2025-01-15:
Sensor types discussed in meeting 26/6/24.
* See Discord for info. 
* Add to the "Competition Entry" Word doc on the OneDrive under the Body > Analog and Sensor Requirements.

EGoliath, 2025-01-19:
https://unisydneyedu-my.sharepoint.com/:w:/g/personal/lwat0045_uni_sydney_edu_au/EY3tUiBB1OVOr_8obefjqW4B51tYOqsHgOD0_Yqsru4BOQ?e=FhOIxb

moiseskim98, 2025-02-15:
The document that is currently under work is [https://unisydneyedu-my.sharepoint.com/:w:/g/personal/lwat0045_alumni_sydney_edu_au/EW3dMYCGrS5BoNUJLPtlWuYB4M3Je95smwskv35AX_L_Bg?e=kbUzM4](link)

It includes the theory of the principle used by the sensor, high level specs, and the proposed design for this project

closed 2025-09-29 by lewisMW