lewisMW/agriculture-SoC #24: Investigate M0 sleep features

lewisMW, 2025-02-16:
- [ ] Read the ARM docs on [M0 power management](https://developer.arm.com/documentation/dui0497/a/the-cortex-m0-processor/power-management) to get an idea of what Cortex M0 features we can use to optimise for a low-power design.
- [ ] Determine the feasibility of our design being able to use the _**deep sleep**_ state and what we need to do to use it.

Kezhipu, 2025-02-22:
feasibility of our design being able to use the deep sleep state：
Based on our agricultural SoCs for sensing and monitoring of field environments, it is important to balance sensing accuracy with power consumption to meet long endurance.And sensors are often “event” driven : when a sensor detects an abnormality (sudden drop in humidity, gas concentration exceeding the limit, etc.), the SoC needs to wake up in time to handle the situation. Most of the time, it can be put to sleep to save energy.
From Doc, we know that the Cortex-M0 core supports two low-power states: Sleep and Deep Sleep：
Deep Sleep can further turn off the system clock and some peripherals such as PLL and Flash memory, which is more power efficient than normal Sleep.
**But from the discussion with Lewis last week, we may need to keep some of the clocks off. (basically feasible)**
And in Deep Sleep, traditional interrupts cannot directly wake up the processor because the main clock may be turned off; while WIC (Wakeup Interrupt Controller) monitors the interrupt line and quickly restores the system clock and wakes up the kernel when it detects a valid signal. This helps to significantly reduce standby power consumption while still maintaining real-time response to external sensors or events. So WIC is a must.

Kezhipu, 2025-02-22:
Entering Deep Sleep Mode

**1. Setting the SLEEPDEEP bit：**
    Before entering Deep Sleep, you need to set the SLEEPDEEP bit in the System Control Register (SCB->SCR).
**2. Execute the sleep command**
    **__WFI()** (Wait For Interrupt) or **__WFE()** (Wait For Event) can be selected：
    __WFI() (Wait For Interrupt): Immediately goes to sleep after execution and waits for interrupt to wake up.
    __WFE() (Wait For Event): if there is no “event” flag, then sleep, otherwise execution continues.
Personally, I think our best choice is **WFI** because it is the most energy efficient, completely stopping until an external interrupt is triggered. And it can be used in conjunction with WIC, and as we can see later, the way we wake up also depends entirely on the way we go to sleep!
   3.Of course, in order to further reduce power consumption, peripherals that are not relevant to the current stage can be turned off by the clock controller or a lower frequency clock source can be used. For example we can keep only the low power timers or critical paths like GPIO input detection.

Kezhipu, 2025-02-22:
Wake Up from Deep Sleep Mode:

As I said before, the mothed of waking up depends on how do we enter deep sleep:
Since we are using WFI, and since the Cortex-M0 shuts down system clocks in deep sleep, traditional interrupt detection won’t work. We have to use **WIC** and even without needing the main clock!

closed 2025-03-01 by lewisMW